slyglif/powerwall3mqtt #14: Exception if host is ureachable or too many retries resulting in crash

ehendrix23, 2025-02-26:
When there is an exception of too my retries due to host unreachable or so then instead of handling the exception the add-on crashes.
This is with version 0.1.1

Following are some of the log file entries:

```
23:01:12] INFO: Preparing to start...
[23:01:13] INFO: Starting Powerwall3MQTT...
2025-02-25T230114.221 - __main__ - DEBUG - Runtime config:
2025-02-25T230114.221 - __main__ - DEBUG - config['log_level'] = DEBUG
2025-02-25T230114.222 - __main__ - DEBUG - config['mqtt_base_topic'] = homeassistant
2025-02-25T230114.222 - __main__ - DEBUG - config['mqtt_ca'] = None
2025-02-25T230114.222 - __main__ - DEBUG - config['mqtt_cert'] = None
2025-02-25T230114.222 - __main__ - DEBUG - config['mqtt_host'] = core-mosquitto
2025-02-25T230114.223 - __main__ - DEBUG - config['mqtt_key'] = None
2025-02-25T230114.223 - __main__ - DEBUG - config['mqtt_password'] = <password>
2025-02-25T230114.223 - __main__ - DEBUG - config['mqtt_port'] = 1883
2025-02-25T230114.223 - __main__ - DEBUG - config['mqtt_ssl'] = False
2025-02-25T230114.224 - __main__ - DEBUG - config['mqtt_username'] = addons
2025-02-25T230114.224 - __main__ - DEBUG - config['mqtt_verify_tls'] = False
2025-02-25T230114.224 - __main__ - DEBUG - config['tedapi_password'] = <password>
2025-02-25T230114.224 - __main__ - DEBUG - config['tedapi_poll_interval'] = 10
2025-02-25T230114.225 - __main__ - DEBUG - config['tedapi_report_vitals'] = False
2025-02-25T230114.225 - __main__ - DEBUG - MQTT will set on 'powerwall3mqtt-488/will' to 'b'offline''
2025-02-25T230114.228 - pytedapi - DEBUG - Testing Connection to Powerwall Gateway: 192.168.91.1
2025-02-25T230114.846 - pytedapi - ERROR - Unable to connect to Powerwall Gateway 192.168.91.1
2025-02-25T230114.847 - pytedapi - ERROR - Please verify your your host has a route to the Gateway.
2025-02-25T230114.847 - pytedapi - ERROR - Error Details: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fad9e0a10>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2025-02-25T230114.847 - __main__ - ERROR - HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fad9e0a10>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/util/connection.py", line 95, in create_connection
    raise err
  File "/usr/lib/python3.12/site-packages/urllib3/util/connection.py", line 85, in create_connection
    sock.connect(sa)
OSError: [Errno 113] Host is unreachable
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 716, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 404, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 1061, in _validate_conn
    conn.connect()
  File "/usr/lib/python3.12/site-packages/urllib3/connection.py", line 363, in connect
    self.sock = conn = self._new_conn()
                       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connection.py", line 186, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7fad9e0a10>: Failed to establish a new connection: [Errno 113] Host is unreachable
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 667, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 802, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/util/retry.py", line 594, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fad9e0a10>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/powerwall3mqtt.py", line 284, in <module>
    app.run()
  File "/app/powerwall3mqtt.py", line 229, in run
    self.tedapi = pytedapi.TEDAPI(
                  ^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 203, in __init__
    self.connect()
  File "/app/pytedapi/__init__.py", line 216, in connect
    return self.reconnect()
           ^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 235, in reconnect
    resp = requests.get(url, verify=False, timeout=5)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 700, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fad9e0a10>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
```

slyglif, 2025-02-27:
OK, 0.1.2 released.  I also changed the structure in git, so please let me know if HA complains when you try to update.

ehendrix23, 2025-02-27:
No issues when I updated. 
Also, link on info page works now. 😀