slyglif/powerwall3mqtt #3: Energy production / consumption metrics

slyglif, 2025-02-26:
The add-on doesn't currently report energy metrics for consumption and production.  Currently a limitation of pypowerwall / the TEDAPI.

abwino, 2025-03-18:
Hi @slyglif Chris, My 'Battery In' sensor taken from the Battery Power Charge entity is showing 1,934kWh so far today, which would be nice! The actual amount of charge today has been 8.2kWh.

slyglif, 2025-03-18:
wow, that's definitely off.  What settings do you have on the Riemann sum?  I created mine in configuration.yaml with the following:
```
sensor:
  - platform: integration
    name: "Home Battery Energy Charged"
    source: sensor.home_battery_power_charge
    unique_id: home_battery_energy_charged
    method: left
```

abwino, 2025-03-18:
Just the normal left method. It's the first time I've used a helper for this, I normally create sensors in my sensors.yaml. I'll delete the helper and start again the old fashioned way! I'll let you know what I get tomorrow. Thanks.

danmalone326, 2025-03-19:
To include my Powerwall3 data on the Energy dashboard, I had to do the following. My Powerwall 3s were only recently installed, so I did not have this data before, but the energy dashboard seems to accurately reflect what I’m seeing in the Netzero and Tesla apps. I would be interested to see if this works for others.

I created 4 integral helpers, using the GUI, to compute energy sensors based on the included power sensors. I did not create the 5th for the Solar Energy, because I get that value direct from an external inverter, but I assume it would work the same. It would be nice if these sensors were automatically included in the add-in.

- My Home Battery Energy Discharge
- My Home Battery Energy Charge
- My Home Grid Energy Import
- My Home Grid Energy Export
- My Home Solar Energy Produced

Here’s an example setup, the others are the same, just change the name and the input sensor:
Settings -> Devices & Services -> Helpers -> Create Helper -> Integral sensor

- Name: My Home Battery Energy Discharge
- Metric prefix: k (kilo)
- Time unit: Hours
- Input sensor: My Home Battery Power Discharge
- Integration method: Trapezoidal rule (default)
- Precision: (default)
- Max sub-interval: <leave blank> (default) --- Do not set to 0 (zero)

[image]

slyglif, 2025-03-19:
Thanks @danmalone326 for posting that, it will definitely help some set it up.

> It would be nice if these sensors were automatically included in the add-in.

I would definitely set them up automatically if I could figure out how, but so far I can't find a way to define helpers as MQTT entities, which is the only way the add-on actually interacts with entities in HA.

abwino, 2025-03-20:
> wow, that's definitely off. What settings do you have on the Riemann sum? I created mine in configuration.yaml with the following:
>
> ```
> sensor:
>   - platform: integration
>     name: "Home Battery [...]

Ok, I deleted my sensors and started again using an Integral helper and a Utility Meter sensor (for daily roll over) and used the 'battery power charge' entity exposed by Powerwall3 to mqtt.

Now 24 hours have passed I can make comparisons between the amount of battery charge declared by the Tesla app, Teslemetry entities and Powerwall3 to mqtt.

Tesla app & Teslemetry - 10kWh
Powerwall3 to mqtt entity - 10.75kWh

I'm going to change my Integral helper to use the trapezoid method instead of left to see if that makes any difference tomorrow.

slyglif, 2025-04-02:
@abwino just curious what the difference looks like using trapezoid?  Also, is the Tesla app / Teslemetry only giving integers for kWh?  If so that could explain a lot of the difference since it could be 10.49kWh (of course it could also be worse and actually be 9.5kWh).

abwino, 2025-04-03:
> [@abwino](https://github.com/abwino) just curious what the difference looks like using trapezoid? Also, is the Tesla app / Teslemetry only giving integers for kWh? If so that could explain a lot of the difference since it could be 10.49kWh (of course it could also be worse and actually be 9.5kWh).

Hi - totally forgot! I switched the template to Trapezoidal last night and the figures from solar charging so far today are:

Tesla App: 2.4kWh
Riemann Trapezoidal:  2.7kWh
Teslemetry: 2.65kWh

Attached are the sensors available through Teslemetry

I'll update this tonight with another reading.

[image]

carbon198, 2025-04-03:
I swapped mine to Trapezoidal and it seems to have retroactively made the numbers match my Tesla app within +/- 0.2kWh.

abwino, 2025-04-03:
> [@abwino](https://github.com/abwino) just curious what the difference looks like using trapezoid? Also, is the Tesla app / Teslemetry only giving integers for kWh? If so that could explain a lot of the difference since it could be 10.49kWh (of course it could also be worse and actually be 9.5kWh).

Here's the update 6 hours later now the sun has gone down:

Tesla App: 10.2 kWh
Riemann Trapezoidal: 12 kWh (read from the energy dashboard)
Teslemetry: 10.53 kWh

It looks like I'll be paying my Teslemetry sub for the foreseeable future! I'm a wine expert by trade, not a coder or maths whizz, so I'm hoping someone can explain why we see these differences in the energy dashboard. Here's the code for my sensor/template, I think it's ok.

In my Templates.yaml

```
- sensor:
    - name: Powerwall Battery Charge Power
      state_class: measurement
      icon: mdi:battery
      unit_of_measurement: W
      device_class: power
      state: >
        {{ states('sensor.my_home_battery_power_charge')}}
```

In my Sensors.yaml

```
- platform: integration
  name: Powerwall Battery Charge Energy kWh
  unique_id: powerwall_charge_energy
  source: sensor.powerwall_battery_charge_power
  unit_prefix: k
  unit_time: h
  method: left
```
:)

slyglif, 2025-04-03:
Tesla app vs Teslemetry is definitely odd, since Teslemetry uses the Fleet API and gets the data direct from Tesla, so it should match.

Riemann sums is most likely because of the polling interval.  I believe Tesla is doing 0.5s inside the PW, but we can't do it that fast without the PW cutting us off.

ehendrix23, 2025-07-18:
I was checking if anything had been changed to be able to get the energy values from TEDAPI by now. :-)
As I was doing, I stumbled on another python library that also uses TEDAPI and they seem to have figured out there how to get the energy intervals I believe.
https://github.com/jasonacox/pypowerwall

If I look in the [__init__.py](https://github.com/jasonacox/pypowerwall/blob/main/pypowerwall/tedapi/__init__.py) I can see what seems to be them getting the lifetime values for energy.

In def vitals I can see entries like:
METER_X_LifetimeEnergyExport": lookup(sync, ['METER_X_AcMeasurements', 'METER_X_LifetimeEnergyExport']),
            "METER_X_LifetimeEnergyImport": lookup(sync, ['METER_X_AcMeasurements', 'METER_X_LifetimeEnergyImport']),

Haven't been able to look further into it and compare it with how it is done here, but figured mentioning it. :-)