slyglif/powerwall3mqtt #5: Ability to control backup reserve percentage

slyglif, 2025-02-26:
Need to add the ability to change the backup reserve percentage

slyglif, 2025-06-28:
@bjquinn thank you for the offer.  Unfortunately the problem isn't with MQTT, it's actually needing to reverse engineer Tesla's  direct API, which so far nobody has been able to do (NetZero talks to Tesla via the cloud based Fleet API).  The best chance at doing so is trying to proxy all traffic from the phone through some system you control, spoofing their certificates, and hoping you can see all the required traffic.  Based on how their app works, it looks like the app has to authenticate to Tesla's cloud, get a token, and then can use the token to communicate with the PW.  At this point we don't know what API the app uses for the authentication or if the API the Tesla app uses to talk to the PW is the same as what we are using to read the stats.

bjquinn, 2025-06-28:
Well that stinks.  I had previously messed around with the Tesla Fleet API, maybe I'll try that again for just this one automation task (setting the backup reserve percentage).

I do want to add that I very much appreciate this powerwall3mqtt add-on, thank you very much for developing it and making it public!  I've been able to fully automate my solar/HVAC/battery with this add-on as the core piece.