slyglif/powerwall3mqtt #17: Unable to connect to Powerwall Gateway 192.168.91.1 (located in UK, PW3 and GW2)

wyx087, 2025-02-27:
Add-on version:  0.1.2 

I've followed your README with adding PING in HA and creating the static route. 

[image]

But I'm still getting the connection error. 

```[15:48:50] INFO: Preparing to start...
[15:48:50] INFO: Starting Powerwall3MQTT...
2025-02-27T154851.177 - __main__ - DEBUG - Runtime config:
2025-02-27T154851.177 - __main__ - DEBUG - config['log_level'] = DEBUG
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_base_topic'] = homeassistant
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_ca'] = None
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_cert'] = None
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_host'] = core-mosquitto
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_key'] = None
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_password'] = tucahhaixah0theceic8Wie8Wohteichoo2veiz7ohQuaaG4eife9Equahr1Ue5Z
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_port'] = 1883
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_ssl'] = False
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_username'] = addons
2025-02-27T154851.177 - __main__ - DEBUG - config['mqtt_verify_tls'] = False
2025-02-27T154851.177 - __main__ - DEBUG - config['tedapi_host'] = 192.168.91.1
2025-02-27T154851.177 - __main__ - DEBUG - config['tedapi_password'] = KXJAP
2025-02-27T154851.177 - __main__ - DEBUG - config['tedapi_poll_interval'] = 10
2025-02-27T154851.177 - __main__ - DEBUG - config['tedapi_report_vitals'] = False
2025-02-27T154851.177 - __main__ - DEBUG - MQTT will set on 'powerwall3mqtt-774/will' to 'b'offline''
2025-02-27T154851.178 - pytedapi - DEBUG - Testing Connection to Powerwall Gateway: 192.168.91.1
2025-02-27T154851.194 - pytedapi - DEBUG - Detected Powerwall 3 Gateway
2025-02-27T154851.194 - pytedapi - DEBUG - Fetching din from Powerwall...
2025-02-27T154851.212 - pytedapi - ERROR - Unable to connect to Powerwall Gateway 192.168.91.1
2025-02-27T154851.212 - pytedapi - ERROR - Please verify your your host has a route to the Gateway.
2025-02-27T154851.212 - pytedapi - ERROR - Error Details: name 'TEDAPIAccessDeniedException' is not defined
2025-02-27T154851.212 - __main__ - ERROR - name 'TEDAPIAccessDeniedException' is not defined
Traceback (most recent call last):
  File "/app/powerwall3mqtt.py", line 293, in <module>
    app.run()
  File "/app/powerwall3mqtt.py", line 235, in run
    self.tedapi = pytedapi.TEDAPI(
                  ^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 203, in __init__
    self.connect()
  File "/app/pytedapi/__init__.py", line 216, in connect
    return self.reconnect()
           ^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 240, in reconnect
    return self.get_din()
           ^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 336, in get_din
    r = self.request("tedapi/din", force=force)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 285, in request
    self.checkHttpResponse(r)
  File "/app/pytedapi/__init__.py", line 255, in checkHttpResponse
    raise TEDAPIAccessDeniedException()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
NameError: name 'TEDAPIAccessDeniedException' is not defined
```

I'm not sure what to do next....... 

Should HA native integration work after creating static route? 

[image]

slyglif, 2025-02-27:
I've got a GW3, so I'm guessing a bit on how it works with a GW2, but my understanding is when using a PW3 you always connect to the PW3 leader, not the gateway.  If my belief is correct, you should be setting the route to point at the PW3, not the gateway, and using the PW3's password instead of the GW's password (the PW3's password is printed on a sticker on it right below the wifi name, which is something like TeslaPW_something).  If I'm incorrect, then you shouldn't need to use the static route and should just be able to use the native integration to talk to the GW's IP instead.

wyx087, 2025-02-27:
Thanks for looking at this. 

This seems like connection issue rather than password issue?? 

My PW3 is wired with a local network IP. Trying to add using local network IP via HA’s native Powerwall integration also returns above, same error. Is this MQTT bridge using the same entry point to read from Powerwall? 

There is no gateway IP that I know of. There is only 1 Powerwall IP in my router connections list and I’m failing to connect to it with everything I’ve tried.

slyglif, 2025-02-27:
The native integration doesn't work the same way.  It talks to a gateway and doesn't need the special route.

So it's likely the route is correct, and now it's just a password issue.  The password for the PW3 is printed on it (or in it, not exactly sure, and I won't be at my Powerwalls for another month or so). I redacted it out of the bug report (and opened a new bug report to redact the passwords in the logs), but the password you had pasted in was only 5 characters, and it definitely needs to be the full password printed on the label of the PW3.