slyglif/powerwall3mqtt #22: Addon stopped working. Error: Read timed out. (read timeout=5)

meow4agreencard, 2025-03-01:
Now running 0.1.3, but the same error happened with 0.0.6 last night.

The add-on stopped working around 2.40 am local time. The following errors appear after startup. Full logs in the txt file show that there is some data being sent from PW3. 
[add-on startup log.txt](https://github.com/user-attachments/files/19040400/add-on.startup.log.txt)
I double-check and Netzero can still connect to the PW3.
Ping from home assistant to 192.168.91.1 works fine.

`2025-03-01T125433.584 - __main__ - ERROR - HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 468, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 463, in _make_request
    httplib_response = conn.getresponse()
                       ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/http/client.py", line 1430, in getresponse
    response.begin()
  File "/usr/lib/python3.12/http/client.py", line 331, in begin
    version, status, reason = self._read_status()
                              ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/http/client.py", line 292, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/socket.py", line 720, in readinto
    return self._sock.recv_into(b)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/ssl.py", line 1251, in recv_into
    return self.read(nbytes, buffer)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/ssl.py", line 1103, in read
    return self._sslobj.read(len, buffer)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TimeoutError: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 667, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 802, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/util/retry.py", line 552, in increment
    raise six.reraise(type(error), error, _stacktrace)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/packages/six.py", line 770, in reraise
    raise value
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 716, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 470, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 358, in _raise_timeout
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/powerwall3mqtt.py", line 298, in <module>
    app.run()
  File "/app/powerwall3mqtt.py", line 251, in run
    self.tesla = hamqtt.devices.TeslaSystem(self.mqtt_base_topic, self.tedapi, self.tedapi_report_vitals)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/hamqtt/devices.py", line 167, in __init__
    vitals = tedapi.get_pw3_vitals()
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 908, in get_pw3_vitals
    data = self.get_pw_vitals(pw_din, force)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 829, in get_pw_vitals
    r = self.post(
        ^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 310, in post
    r = requests.post(url,
        ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 115, in post
    return request("post", url, data=data, json=json, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 713, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
`

slyglif, 2025-03-01:
It's interesting that this was on startup.  Was it working prior?  If so, can you get me the log of the last few lines before it crashed before this?

I do (currently) expect it to die if it encounters a problem on startup, but it's odd that it made it all the way to "get_pw_vitals, because that's the 5th major API call it makes.  Could you describe your network a little?  Is it a simple flat network, or do you use different VLANs?  If you're using different VLANs, do you also have firewall rules setup?

meow4agreencard, 2025-03-01:
Let me troubleshoot things a little bit more on my side before I bother you again. I just noticed that the PW3 is only getting power from one string, which is odd because I have 5 connected. Looks like this may be some internal issue with the PW.
I'll check with Tesla support first and get back here after I figure things out.

slyglif, 2025-03-01:
I was just finishing up a big cleanup change on the code, and the cleanup made it easy to add better handling of the case where the vitals call fails (which apparently can happen when some of PV string metering components have issues).  New version v0.2.0 should just keep running, but mark the individual powerwalls (as opposed to the system) as unavailable in HA.  Since I haven't run into this I wasn't able to test it, but please let me know if it works even before Tesla fixes whatever is up with the PW.

meow4agreencard, 2025-03-01:
Looks like one of the PW3s is completely down.

slyglif, 2025-03-01:
That definitely sucks.  I'll take a look at the logs you posted to see if there is something in it I could use to show that kind of status info.  Doesn't help your current situation, but could make it so you could be alerted in the future.

meow4agreencard, 2025-03-01:
> That definitely sucks. I'll take a look at the logs you posted to see if there is something in it I could use to show that kind of status info. Doesn't help your current situation, but could make it so you could be alerted in the future.

Actually, just refreshed the add-on store and upgraded to 0.2. The add-on works again. Thank you!

Now just the minor matter of the dead PW3.

slyglif, 2025-03-01:
Great, glad it works!  Hopefully Tesla will get your dead battery working soon.  In the mean time I am seeing some things in the logs that I can try to expose somehow.  Even though there are 2 batteries, and both show "battery_power_status" = "on", the full pack energy doesn't total to 2 batteries worth (would have to build assumptions about battery sizes in for this to matter), there is a "BatteryComms" alert (definitely something I'll try to expose), and several other specific alerts as well (I've included them below in case it helps you dealing with Tesla, but they should have all of this):

BMS alerts: BMS_a037_SW_NVRAM_Config_Error, BMS_a048_SW_HVP_HVI_Comms, BMS_a083_Nvram_Filecache
pws alerts: PWS_a002_MciSelfTestAB, PWS_a003_MciSelfTestCD, PWS_a004_MciSelfTestEF

slyglif, 2025-03-02:
I just pushed 0.3.0, which adds some additional sensors related to alerts from the power wall.  Please let me know if it shows the Battery Missing Alert as active.

meow4agreencard, 2025-03-02:
Got both Comms and Battery Missing active after upgrading.

slyglif, 2025-03-02:
Great, thank you!  I'm going to close this issue out since it's a hardware problem and I added code so the add-on will keep working.

closed 2025-03-02 by slyglif