slyglif/powerwall3mqtt #23: MQTT connection info not set

chinswain, 2025-03-08:
I've setup a static route and confirmed it's connection via ping, but I can't get past this error (MQTT details are correct). I'm using the EMQX addon rather than Mosquitto.

```
[00:51:19] INFO: Preparing to start...
[00:51:19] ERROR: Got unexpected response from the API: Service not enabled
[00:51:19] INFO: Starting Powerwall3MQTT...
2025-03-08T005119.784 - __main__ - CRITICAL - Exiting: MQTT connection info not set
```

Config
```
tedapi_report_vitals: false
tedapi_poll_interval: 30
mqtt_base_topic: homeassistant
mqtt_ssl: false
tedapi_password: xxxxxx
mqtt_server: xxxxxx
mqtt_port: 1883
log_level: DEBUG
mqtt_password: xxxxxx
mqtt_user: xxxxxx

```
Current version: 0.3.0

slyglif, 2025-03-08:
Thank you for the report.  I'm not able to get on my computer until Sunday, but it looks like an easy fix, so expect 0.3.1 sometime on Sunday.

chinswain, 2025-03-08:
What's the issue? I couldn't work it out.

slyglif, 2025-03-08:
I mixed up the name of the mqtt host and username variables in the config file vs. in the application code, so the app isn't picking up the correct mqtt host name or user.  It's a non-issue when using the Mosquitto add-on, but if someone sets them in the config directly it wouldn't work.  It should be printing those out since debug logging is enabled.  Please let me know as confirmation in case there is another issue I need to look at as well.

slyglif, 2025-03-09:
I've pushed an update.  After installing you'll need to update the config for the add-on to make sure the username and hostname are correct for MQTT.

chinswain, 2025-03-10:
Perfect, that worked!

closed 2025-03-10 by chinswain