slyglif/powerwall3mqtt #26: Device not being being created under MQTT Integration

HEAT16, 2025-03-26:
I've got the routing set up and I've confirmed that powerwall3mqtt is able to see and talk to my powerwall3. I confirmed that Mosquitto broker is seeing the connection from powerwall3mqtt and I confirmed that the MQTT integration is able to see the json values posted to homeassistant/device/# topic. For the life of me I can not get the MQTT integration to show the actual powerwall3 device. I have also confirmed with MQTT Explorer that the config and status messages are present. Not sure what I'm doing wrong.

Mosquitto Log:
2025-03-26 13:49:09: New connection from 192.168.1.243:53162 on port 1883.
2025-03-26 13:49:09: New client connected from 192.168.1.243:53162 as mqtt-explorer-e60a62e1 (p2, c1, k60, u'mqtt').

MQTT Integration Topic Listening: 
{"dev": {"ids": "TeslaEnergySystem_1707000-11-L--TG125006002JHX", "name": "My Home", "mf": "Tesla", "mdl": "Powerwall3", "mdl_id": "1707000-11-L", "sw": "25.2.4 4646ca37", "sn": "TG125006002JHX"}, "state_topic": "homeassistant/device/TeslaEnergySystem_1707000-11-L--TG125006002JHX/state", "availability": [{"topic": "homeassistant/device/TeslaEnergySystem_1707000-11-L--TG125006002JHX/state", "value_template": "{{ value_json.mqtt_availability }}"}, {"topic": "powerwall3mqtt-27/will"}], "o": {"name": "powerwall3mqtt"}, "cmps": {"backfeed_limited": {"p": "binary_sensor", "value_template": "{{ value_json.backfeed_limited }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_backfeed_limited_alert", "name": "Backfeed Limited Alert"}, "battery": {"p": "sensor", "value_template": "{{ value_json.battery }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery", "name": "Battery", "device_class": "battery", "unit_of_measurement": "%"}, "battery_capacity": {"p": "sensor", "value_template": "{{ value_json.battery_capacity }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_capacity", "name": "Battery Capacity", "device_class": "energy_storage", "unit_of_measurement": "Wh"}, "battery_comms": {"p": "binary_sensor", "value_template": "{{ value_json.battery_comms }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_comms_alert", "name": "Battery Comms Alert", "device_class": "problem"}, "battery_missing": {"p": "binary_sensor", "value_template": "{{ value_json.battery_missing }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_missing_alert", "name": "Battery Missing Alert", "device_class": "problem"}, "battery_power": {"p": "sensor", "value_template": "{{ value_json.battery_power }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_power", "name": "Battery Power", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement"}, "battery_remaining": {"p": "sensor", "value_template": "{{ value_json.battery_remaining }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_remaining", "name": "Battery Remaining", "device_class": "energy_storage", "unit_of_measurement": "Wh"}, "battery_reserve_hidden": {"p": "sensor", "value_template": "{{ value_json.battery_reserve_hidden }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_hidden_reserve", "name": "Battery Hidden Reserve", "device_class": "energy_storage", "unit_of_measurement": "Wh"}, "battery_reserve_user": {"p": "sensor", "value_template": "{{ value_json.battery_reserve_user }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_reserve", "name": "Battery Reserve", "device_class": "battery", "unit_of_measurement": "%"}, "battery_time_remaining": {"p": "sensor", "value_template": "{{ value_json.battery_time_remaining }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_time_remaining", "name": "Battery Time Remaining", "device_class": "duration", "unit_of_measurement": "s"}, "calibration": {"p": "binary_sensor", "value_template": "{{ value_json.calibration }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_calibration", "name": "Calibration", "device_class": "running"}, "commission_date": {"p": "sensor", "value_template": "{{ value_json.commission_date }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_commission_date", "name": "Commission Date", "device_class": "timestamp"}, "grid_power": {"p": "sensor", "value_template": "{{ value_json.grid_power }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_grid_power", "name": "Grid Power", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement"}, "grid_status": {"p": "binary_sensor", "value_template": "{{ value_json.grid_status }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_grid_status", "name": "Grid Status", "device_class": "connectivity"}, "inverter_capacity": {"p": "sensor", "value_template": "{{ value_json.inverter_capacity }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_inverter_capacity", "name": "Inverter Capacity", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement"}, "load_power": {"p": "sensor", "value_template": "{{ value_json.load_power }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_load_power", "name": "Load Power", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement"}, "solar_power": {"p": "sensor", "value_template": "{{ value_json.solar_power }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_solar_power", "name": "Solar Power", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement"}, "real_power_config_limited": {"p": "binary_sensor", "value_template": "{{ value_json.real_power_config_limited }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_real_power_config_limited_alert", "name": "Real Power Config Limited Alert"}, "battery_power_in": {"p": "sensor", "value_template": "{{ [ value_json.battery_power | int, 0 ] | min | abs }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_power_charge", "name": "Battery Power Charge", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement", "en": "false"}, "battery_power_out": {"p": "sensor", "value_template": "{{ [ value_json.battery_power | int, 0 ] | max }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_battery_power_discharge", "name": "Battery Power Discharge", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement", "en": "false"}, "grid_power_in": {"p": "sensor", "value_template": "{{ [ value_json.grid_power | int, 0 ] | max }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_grid_power_import", "name": "Grid Power Import", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement", "en": "false"}, "grid_power_out": {"p": "sensor", "value_template": "{{ [ value_json.grid_power | int, 0 ] | min | abs }}", "unique_id": "TeslaEnergySystem_1707000-11-L--TG125006002JHX_grid_power_export", "name": "Grid Power Export", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement", "en": "false"}}}

[e5f48848_powerwall3mqtt_2025-03-26T17-45-46.023Z.log](https://github.com/user-attachments/files/19473152/e5f48848_powerwall3mqtt_2025-03-26T17-45-46.023Z.log)

{"dev": {"ids": "Powerwall3_1707000-11-L--TG125006002JHX", "name": "My Home TG125006002JHX", "via_device": "TeslaEnergySystem_1707000-11-L--TG125006002JHX", "mf": "Tesla", "mdl": "Powerwall3", "mdl_id": "1707000-11-L", "sn": "TG125006002JHX"}, "state_topic": "homeassistant/device/Powerwall3_1707000-11-L--TG125006002JHX/state", "availability": [{"topic": "homeassistant/device/Powerwall3_1707000-11-L--TG125006002JHX/state", "value_template": "{{ value_json.mqtt_availability }}"}, {"topic": "powerwall3mqtt-27/will"}], "o": {"name": "powerwall3mqtt"}, "cmps": {"battery_capacity": {"p": "sensor", "value_template": "{{ value_json.battery_capacity }}", "unique_id": "Powerwall3_1707000-11-L--TG125006002JHX_battery_capacity", "name": "Battery Capacity", "device_class": "energy_storage", "unit_of_measurement": "Wh"}, "battery_remaining": {"p": "sensor", "value_template": "{{ value_json.battery_remaining }}", "unique_id": "Powerwall3_1707000-11-L--TG125006002JHX_battery_remaining", "name": "Battery Remaining", "device_class": "energy_storage", "unit_of_measurement": "Wh"}, "string_A_mode": {"p": "sensor", "value_template": "{{ value_json.strings['A'].mode }}", "unique_id": "Powerwall3_1707000-11-L--TG125006002JHX_pv_string_a_mode",  [... cut ...] on.strings['F'].voltage }}", "unique_id": "Powerwall3_1707000-11-L--TG125006002JHX_pv_string_f_voltage", "name": "PV String F Voltage", "device_class": "voltage", "unit_of_measurement": "V", "en": "false"}, "string_F_power": {"p": "sensor", "value_template": "{{ value_json.strings['F'].power }}", "unique_id": "Powerwall3_1707000-11-L--TG125006002JHX_pv_string_f_power", "name": "PV String F Power", "device_class": "power", "unit_of_measurement": "W", "state_class": "measurement", "en": "false"}}}

slyglif, 2025-03-28:
That looks like the messages are making it to the integration.  Is it definitely not showing up under Settings, Integrations, MQTT, then the devices list under Mosquito broker?  If not, could you verify the settings for the MQTT integration?  Also if not, try enabling debug logging on the MQTT integration and then restart the add-on.

HEAT16, 2025-03-28:
It's for sure not showing up as a device under the MQTT integration.  I added a new shade last night via zigbee2MQTT and it popped right up as a device under the MQTT integration, so I know discovery is working. I verified in the MQTT integration settings that discovery is enabled and that the MQTT username and password are correct. 

Inside the integration when I listen to the topic "homeassistant/device/TeslaEnergySystem_1707000-11-L--TG125006002JHX/#" I can see the messages coming in every 30 seconds, so I know both the Mosquitto broker (add-on) is getting the messages from powerwall3mqtt via MQTT and I know the Integration is properly connecting to the broker because I am able to listen for the topic and see the incoming messages from the broker. 

I do have debugging enabled on the Mosquitto broker but the only messages displayed are like the ones I included in my initial post: New connection from IP address & New Client Connected. 

I manually created the 6 sensor entities needed for the energy dashboard and those seem to be working fine so the only thing I can think of is an issue with the configuration message causing the device to not be auto-discovered.

slyglif, 2025-03-28:
OK, definitely turn on the debug logging on the MQTT integration.  That might give us some extra info on why it isn't working.

HEAT16, 2025-03-28:
I removed everything from before I enabled debugging on the Integration. I looked and I don't see the config message coming through, but the logs from the powerwall3mqtt show it being sent AND I can see it was sent in MQTT Explorer.

powerwall3mqtt log excerpt:
[11:54:02] INFO: Preparing to start...
[11:54:03] INFO: Starting Powerwall3MQTT...
2025-03-28T115404.910 - __main__ - INFO - Powerwall firmware version = 25.2.6 5db0a0f5
2025-03-28T115404.912 - __main__ - INFO - Connected to MQTT Broker '192.168.50.124:1883'
2025-03-28T115404.914 - __main__ - INFO - Discovery sent to 'homeassistant/device/TeslaEnergySystem_1707000-11-L--TG125006002JHX/config'
2025-03-28T115404.916 - __main__ - INFO - Discovery sent to 'homeassistant/device/Powerwall3_1707000-11-L--TG125006002JHX/config'
2025-03-28T115404.916 - __main__ - INFO - Sleeping 0.5s to allow HA to process discovery
2025-03-28T115404.917 - __main__ - INFO - Subscribed to MQTT topic 'homeassistant/status'
2025-03-28T115405.418 - __main__ - INFO - Sent message to 'homeassistant/device/TeslaEnergySystem_1707000-11-L--TG125006002JHX/state'

MQTT Explorer:

[image]

Integration log:
[home-assistant_mqtt_2025-03-28T16-01-12.543Z.log](https://github.com/user-attachments/files/19508840/home-assistant_mqtt_2025-03-28T16-01-12.543Z.log)

slyglif, 2025-03-28:
Try enabling debug logging on the MQTT integration, then restarting the add-on so we can see what the integration thinks of the config message.

slyglif, 2025-03-28:
OK, something is very weird.  I just did it on mine to see the message.  For your power wall, you should see the following message in the log.

```
DATE_TIME DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homeassistant/device/TeslaEnergySystem_1707000-11-L--TG125006002JHX/config (qos=0): b'{"dev": ...
```

I'm not seeing any messages for /config in your log.  Could you verify what your discovery prefix is set to?  To get to it you go to Settings, MQTT Integration, Configure, Configure MQTT Options.  The default is "homeassistant".  It looks like the homeassistant tree is in MQTT, but not seeing the integration pick up the discovery message makes me wonder.

HEAT16, 2025-03-28:
I know, I searched the logs for /config and nothing came up. The extra crazy part is that MQTT explorer sees the config message at 11:54 which you can see in my screenshot so we know the broker is receiving the message. 

I have discovery on in the MQTT Integration: 

[image]

HEAT16, 2025-03-28:
I'm also seeing the config message when I use the MQTT integration to listen for that specific topic:

[image]

To capture this I started listening to the topic on the MQTT Integration, and then restarted the powerwall3mqtt Add-on.

slyglif, 2025-03-28:
That's great that it was an easy fix!  It looks like device based discovery messages were added in 2024.11.0, so I'll update the add-on to require that as a minimum version.

HEAT16, 2025-03-28:
Sorry for all the trouble and thank you for your help!

closed 2025-04-02 by slyglif