slyglif/powerwall3mqtt #27: PV String Sensors showing Unavailable

arothian, 2025-03-29:
Hello!

I recently setup this add-on, and I'm getting good overall Sensor information through MQTT into HA. However, none of the `PV String ...` Sensors show any information. 

I only have one PW3, and it shows two PW3 entities in MQTT. One has the power usage information sensors, which is working, and then the other has the PV sensors that show unavailable.

Am I missing some configuration it get these values?

Let me know what debug information might be useful.

slyglif, 2025-03-29:
One of the devices is for the system as a whole, which is always enabled.  The other is for per-PW data.  To get that data you need to enable "Report Powerwall Vitals" in the add-on configuration and restart.

arothian, 2025-03-29:
Missed that config, thanks!

closed 2025-03-29 by arothian