slyglif/powerwall3mqtt #32: Gateway password

Fenix-HA, 2025-04-03:
Since this mornong I cant't connect to PW3 as always. This is the log 

`
2025-04-03T095508.237 - __main__ - ERROR - (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')
Traceback (most recent call last):
  File "/app/powerwall3mqtt.py", line 348, in <module>
    app.run()
  File "/app/powerwall3mqtt.py", line 288, in run
    tedapi = pytedapi.TeslaEnergyDeviceAPI(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 129, in __init__
    self.connect()
  File "/app/pytedapi/__init__.py", line 147, in connect
    self.reconnect()
  File "/app/pytedapi/__init__.py", line 169, in reconnect
    self.get_din(force=True)
  File "/app/pytedapi/__init__.py", line 267, in get_din
    r = self.request("tedapi/din", force=force)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 216, in request
    self.check_http_response(r)
  File "/app/pytedapi/__init__.py", line 186, in check_http_response
    raise exceptions.TEDAPIAccessDeniedException()
pytedapi.exceptions.TEDAPIAccessDeniedException: (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')
`
The password in config is the same that I always been using and the same for connecting to PW3 WiFi

shellhouse4, 2025-04-03:
I'm encountering the same today.  Everything was working up until today.

[10:18:30] INFO: Preparing to start...
[10:18:31] INFO: Starting Powerwall3MQTT...
2025-04-03T101833.523 - __main__ - DEBUG - Runtime config:
2025-04-03T101833.523 - __main__ - DEBUG - config['log_level'] = 'DEBUG'
2025-04-03T101833.524 - __main__ - DEBUG - config['mqtt_base_topic'] = 'homeassistant'
2025-04-03T101833.525 - __main__ - DEBUG - config['mqtt_ca'] = 'None'
2025-04-03T101833.525 - __main__ - DEBUG - config['mqtt_cert'] = 'None'
2025-04-03T101833.526 - __main__ - DEBUG - config['mqtt_host'] = 'core-mosquitto'
2025-04-03T101833.526 - __main__ - DEBUG - config['mqtt_key'] = 'None'
2025-04-03T101833.528 - __main__ - DEBUG - config['mqtt_password'] = 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX'
2025-04-03T101833.528 - __main__ - DEBUG - config['mqtt_port'] = '1883'
2025-04-03T101833.529 - __main__ - DEBUG - config['mqtt_ssl'] = 'False'
2025-04-03T101833.529 - __main__ - DEBUG - config['mqtt_username'] = 'addons'
2025-04-03T101833.530 - __main__ - DEBUG - config['mqtt_verify_tls'] = 'False'
2025-04-03T101833.530 - __main__ - DEBUG - config['tedapi_host'] = '192.168.91.1'
2025-04-03T101833.531 - __main__ - DEBUG - config['tedapi_password'] = 'XXXXXXXXXX'
2025-04-03T101833.532 - __main__ - DEBUG - config['tedapi_poll_interval'] = '30'
2025-04-03T101833.532 - __main__ - DEBUG - config['tedapi_report_vitals'] = 'False'
2025-04-03T101833.534 - __main__ - DEBUG - MQTT will set on 'powerwall3mqtt-916/will' to 'b'offline''
2025-04-03T101833.539 - pytedapi - DEBUG - Testing Connection to Powerwall Gateway: 192.168.91.1
2025-04-03T101834.634 - pytedapi - DEBUG - Detected Powerwall 3 Gateway
2025-04-03T101834.635 - pytedapi - DEBUG - Fetching din from Powerwall...
2025-04-03T101834.766 - __main__ - ERROR - (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')

sbman, 2025-04-03:
Mine is still running OK as of this morning.  What version of the PW firmware is your powerwall running?  You can find this in Settings->Integrations->MQTT->Devices->Your powerwall

Mine says:
Firmware: 25.2.4 4646ca37

shellhouse4, 2025-04-03:
Firmware: 25.2.6 5db0a0f5 on here.  Must have updated.

Fenix-HA, 2025-04-03:
Mine is  25.2.6 5db0a0f5 (?!?!)

sbman, 2025-04-03:
Are either of you on Linux?  If so, can you test this command:
curl --user Tesla_Energy_Device:YOURPOWERWALLPASSWORD -k --insecure https://192.168.91.1/tedapi/din

You'll need to be on the same network as the powerwall3 and have the route added.  Uou should be able to ping 192.168.91.1 before this command will work.  This should return your powerwall's ID number, but I'm expecting it will be returning an error instead.

slyglif, 2025-04-03:
Based on the logs, it's failing on the call to /tedapi/din, so I would expect the curl command to fail as well.  There was a similar bug reported opened on Powerwall-Dashboard this morning (which makes sense, since I based this on the pypowerwall library that PD uses).  Teslemetry, from what I can tell, uses the Fleet API and doesn't talk directly to the PW, so it wouldn't be affected.

I suspect Tesla changed how the PW firmware works, specifically how it handles requests.  More than likely they added a config option to the web server that if the request comes from an IP outside of 192.168.91.0/24 they reject it as not authorized.  If that's the case, fixing this may not be possible (just doing a NAT wouldn't work because the PW would try to send the reply to their wifi network).

The firmware hasn't rolled out to my PWs yet, so I can't see the effect first hand, or try workarounds.  Once it hits mine I'll definitely be trying some other options to get it to communicate.

sbman, 2025-04-03:
If that is what they have done, I am running mine in a stand alone docker container so I can put it on any network I want.  Once this update hits my powerwalls I can try various strategies as well.  It's incredibly frustrating that Tesla doesn't provide a documented, supported local API. I paid a lot for this equipment, I should be able to use it the way I want.

sbman, 2025-04-03:
> Based on the logs, it's failing on the call to /tedapi/din, so I would expect the curl command to fail as well. There was a similar bug reported opened on Powerwall-Dashboard this morning (which [...]

One other thing, supposedly the Tesla app can communicate with the PW directly if your internet connection/wifi is offline.  I would think they need to continue to support that and messing with network source rules would be a problem for that.

slyglif, 2025-04-03:
> If that is what they have done, I am running mine in a stand alone docker container so I can put it on any network I want. Once this update hits my powerwalls I can try various strategies as well. [...]

If you go that route, you'll need to setup your host so it connects to the PW WiFi network (the one listed on the label), and pass that through to the container.  Should be doable, but definitely not clean or easy for most people to do.

Unfortunately that won't be an option for me because I also have a SPAN panel, and both are hardwired.  As of 4-6 weeks ago there is an issue where the SPAN can't communicate with the PW if the PW is on both wired and WiFi, and I suspect that was caused by their first steps to restrict how it handles IP routing.  The fix SPAN had me implement was disconnecting / blocking the PW from my wifi network (which has the unfortunate side effect of the PW not choosing the best WiFi channel for cooperation with the other WiFi networks around it).

I definitely agree with the frustration of them not providing a proper local API.  But they apparently want to turn the data into another profit center for Tesla, so I doubt they'll make something available.

> One other thing, supposedly the Tesla app can communicate with the PW directly if your internet connection/wifi is offline. I would think they need to continue to support that and messing with network source rules would be a problem for that.

If local WiFi is down and it can somehow still communicate with the PW then it either has the user connect to the PW WiFi, or it uses Bluetooth (which I think the PW3s support).  If it can still communicate of the main WiFI then there is definitely something else going on, and it could be some additional authentication they've added.  That could potentially be fixable, but it would likely require capturing the additional auth data for each PW, so it wouldn't be as easy to setup.

Fenix-HA, 2025-04-03:
> Are either of you on Linux? If so, can you test this command: curl --user Tesla_Energy_Device:YOURPOWERWALLPASSWORD -k --insecure https://192.168.91.1/tedapi/din
>
> You'll need to be on the same [...]

I'm on Windows unfortunately

sbman, 2025-04-03:
Mine updated to 25.2.7 and now not able to connect.

sbman, 2025-04-03:
The call to https://192.168.91.1/tedapi/din is still working when I call it from curl, so it's something further down the line.

sbman, 2025-04-03:
Yes, I'm working on it, turns out I'm seeing an MQTT issue because I'm in the middle of updating my docker config to something reasonable that can be pushed to git, not the hack I did in 10 minutes to make it work...  So I haven't seen for sure if PW 25.2.7 is actually a problem or not. The curl call did work OK for the din so maybe it was just an issue with 25.2.6

sbman, 2025-04-03:
I finally got the stand alone container that I've been working on with ENV configuration and running.  On PW 25.2.7 powerwall3mqtt is connecting and running just fine.

slyglif, 2025-04-03:
That's good to hear.  Hopefully .6 was just a bad version and when those that were upgraded to it get .7 all will be resolved.  Unrelated, I plan to work on the other issues / PR you submitted this weekend.

sbman, 2025-04-03:
I commited a set of files in my fork of your repo that has a Dockerfile and docker-compose example with a README for making a standalone container to run with HA Community.  I'm not sure what to do with it beyond that, possibly I could host a docker image somewhere so people don't have to build the image manually.

slyglif, 2025-04-04:
[This thread](https://github.com/jasonacox/Powerwall-Dashboard/discussions/109) over in the Powerwall Dashboard project is not promising.  I suspect Tesla has closed the TEDAPI to anything not coming from 192.168.91.0/24, which would not be possible to use without somehow being on the Tesla hosting WiFi network (since return traffic to that subnet would only get sent out on their WiFi network).

F1p, 2025-04-08:
A bit strange because of the issue title...
Commissioned 9hrs, updated to 25.2.6 5db0a0f5 and this add-on is working (amazing btw - thank you so much)

Question is... Should I block internet connection for the PW to prevent it updating and breaking?

slyglif, 2025-04-08:
It could be a combination of the PW firmware and the gateway firmware, but unknown.  .10 definitely seems to cause problems as well based on an issue on Powerwall-Dashboard.

I wouldn't recommend blocking access, because I believe a standard line in the warranty paperwork for the PW is it must have an internet connection, and if it can't communicate with the mothership they have the right to cancel the warranty.

F1p, 2025-04-08:
Thanks for the quick reply.. hmm perhaps it will need to be taken up with my ISP why they can't repair my broken internet connection because they only work on Feb 29th 🤣

Seriously though.. Yeah I was reading the .10 and permanent connections to the WiFi isn't impossible but a hassle...

So my situation is single PW3 and GW2

slyglif, 2025-04-08:
I'm not sure what causes some to fail.  I've now been upgraded to 25.2.6 and it's still working for me.  We'll see what happened when the .10 upgrade comes around.  My rPi is within range of the PWs, so I can do the wifi workaround at that level, but I don't know if it will cause problems with the SPAN panel (I had to block the PW from my wifi because the SPAN was no longer able to communicate with it properly because the PW was apparently sending response traffic back from the wrong IP).

Fenix-HA, 2025-04-09:
I confirm that the problem is with .10 version, I think I misunderstood as mqtt shows me the old version, but with Tesla app I see the new one

F1p, 2025-04-09:
Thanks for clarifying, guess will start preparing the alternative for when .10 drops in

slyglif, 2025-04-09:
Looks like mine upgraded last night and I'm dead in the water like others.  I'll be able to do the wifi workaround when I'm onsite next time, but it basically means I can't do anything until late next month.

slyglif, 2025-04-09:
I've updated the README to reflect the requirement of using the WiFi network.  Hopefully a different method will be discovered.

Fenix-HA, 2025-04-09:
> I've updated the README to reflect the requirement of using the WiFi network. Hopefully a different method will be discovered.

Sorry for the question but since my HA is in range of PW's WiFi do you know how can I connect/interact with it?

slyglif, 2025-04-09:
> Sorry for the question but since my HA is in range of PW's WiFi do you know how can I connect/interact with it?

Since I can't test this myself I'm guessing blind.  I believe the steps would be:
1. If you've setup the Static Routes Manager add-on, you should remove it and reboot HA.
2. Go to Settings, System, Network
3. Configure the WLAN0 interface
4. Search for Wi-Fi networks
5. Click on the TeslaPW_XXXXX network and configure it to use DHCP
6. Restart the add-on

Since mine is out of range I won't be able to directly test these steps, so if you and others could document the exact process I will update the docs accordingly.

F1p, 2025-04-09:
Above is a good idea, I would need to relocate my proxmox server and play about with wireless adapters...

What about making a standalone docker image of this project in addition to this add-on to make it more portable for e.g. raspi pi install so no need to run full HA?

slyglif, 2025-04-09:
> Above is a good idea, I would need to relocate my proxmox server and play about with wireless adapters...
> 
> What about making a standalone docker image of this project in addition to this add-on to make it more portable for e.g. raspi pi install so no need to run full HA?

I'll need that for myself, so I'll definitely be working on that.  My rPi is out of range, so I'll have to put this on a different server to make it work.  If anyone has some good advice on a case for an rPi 4b + PoE+ HAT, preferably ABS or some other plastic instead of aluminum, I'd appreciate it.

chinswain, 2025-04-09:
Is there any off the self devices that could connect to the Wifi hotspot and bridge it to the home network? What about an ESP32 that connects to the PW3 hotspot and relays the data via another ESP32 that's connected to the home WiFi?

slyglif, 2025-04-09:
There might be, but I don't know of one off hand.  In my case I'm almost definitely going to use a PoE powered rPi running this as a docker container, just because it will be easy to manage.

sbman, 2025-04-09:
I think that if you have networking hardware that can create VLAN and WIFI Networks, you should be able to create a new VLAN and WIFI network that is on 192.168.91.0/24 network, have the PW join that WIFI, then put your 'server' running pypowerwall  on the same VLAN and it should not know the difference.

F1p, 2025-04-09:
> Is there any off the self devices that could connect to the Wifi hotspot and bridge it to the home network? What about an ESP32 that connects to the PW3 hotspot and relays the data via another ESP32 that's connected to the home WiFi?

I'm going to get a PoE hat for a ESP32 to test it, but it would be starting a project from scratch...
If it's wireless you use, then being a dual client of two access points at the same time is a problem (need two wireless radio's)

The Pi + PoE hat sounds the best option at the moment

slyglif, 2025-04-09:
> I think that if you have networking hardware that can create VLAN and WIFI Networks, you should be able to create a new VLAN and WIFI network that is on 192.168.91.0/24 network, have the PW join [...]

Unfortunately it won't quite work.  The PW only uses the 192.168.91.0/24 network on the TeslaPW_XXXXX WiFi network, and it only uses that in AP mode, not client mode.  So it can't join a WiFi network created by another AP.  If you have a WiFi bridge device that can act as a WiFi client, then it could join the PW's WiFi network and bridge it to the VLAN of your choice though.  In my case, UniFi AP's can't act as a client, only an AP, so I'd need another device.

Also, once you have it bridged, you'd need to either have HA on the same VLAN, or have something running NAT rules so the source IP for all comms going to the PW is in the 192.168.91.0/24 and not just an IP from another routed VLAN.

F1p, 2025-04-09:
> There might be, but I don't know of one off hand. In my case I'm almost definitely going to use a PoE powered rPi running this as a docker container, just because it will be easy to manage.

With you, regarding Pi's.. the official Pi5 case and Pi5 POE+ hat fits by the looks of it... the 5 might be a bit overkill though!

sbman, 2025-04-10:
I have an image that is setup for use in standalone.

You can see a sample docker compose file here: [hacore/docker-compose.yaml](https://github.com/sbman/powerwall3mqtt-hacore/blob/d7324140fa62c8c4cbf7601b3ffcd5b715432697/hacore/docker-compose.yaml)

You need to set the proper values for your MQTT, tedapi password, etc. and then use docker compose to bring it up.  I recommend setting the restart to never and running it interactively so you can see any errors before you run it -d to put it in the background.

Fenix-HA, 2025-04-13:
> Since I can't test this myself I'm guessing blind. I believe the steps would be:
>
> 1. If you've setup the Static Routes Manager add-on, you should remove it and reboot HA.
> 2. Go to Settings, System, [...]

So I'v tried and HA is connected through PW WiFi, but nothing change

`[11:04:31] INFO: Preparing to start...
[11:04:34] INFO: Starting Powerwall3MQTT...
2025-04-13T110439.066 - __main__ - DEBUG - Runtime config:
2025-04-13T110439.067 - __main__ - DEBUG - config['log_level'] = 'DEBUG'
2025-04-13T110439.068 - __main__ - DEBUG - config['mqtt_base_topic'] = 'homeassistant'
2025-04-13T110439.069 - __main__ - DEBUG - config['mqtt_ca'] = ''
2025-04-13T110439.070 - __main__ - DEBUG - config['mqtt_cert'] = ''
2025-04-13T110439.070 - __main__ - DEBUG - config['mqtt_host'] = 'core-mosquitto'
2025-04-13T110439.071 - __main__ - DEBUG - config['mqtt_key'] = ''
2025-04-13T110439.073 - __main__ - DEBUG - config['mqtt_password'] = 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX'
2025-04-13T110439.073 - __main__ - DEBUG - config['mqtt_port'] = '1883'
2025-04-13T110439.074 - __main__ - DEBUG - config['mqtt_ssl'] = 'False'
2025-04-13T110439.075 - __main__ - DEBUG - config['mqtt_username'] = 'addons'
2025-04-13T110439.076 - __main__ - DEBUG - config['mqtt_verify_tls'] = 'False'
2025-04-13T110439.076 - __main__ - DEBUG - config['tedapi_host'] = '192.168.91.1'
2025-04-13T110439.077 - __main__ - DEBUG - config['tedapi_password'] = 'XXXXXXXXXX'
2025-04-13T110439.078 - __main__ - DEBUG - config['tedapi_poll_interval'] = '30'
2025-04-13T110439.079 - __main__ - DEBUG - config['tedapi_report_vitals'] = 'False'
2025-04-13T110439.081 - __main__ - DEBUG - MQTT will set on 'powerwall3mqtt-280/will' to 'b'offline''
2025-04-13T110439.231 - pytedapi - DEBUG - Testing Connection to Powerwall Gateway: 192.168.91.1
2025-04-13T110442.323 - pytedapi - DEBUG - Detected Powerwall 3 Gateway
2025-04-13T110442.324 - pytedapi - DEBUG - Fetching din from Powerwall...
2025-04-13T110448.556 - pytedapi - ERROR - Unable to connect to Powerwall Gateway: 192.168.91.1
2025-04-13T110448.556 - pytedapi - ERROR - Please verify your your host has a route to the Gateway.
2025-04-13T110448.557 - __main__ - ERROR - HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 404, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 1061, in _validate_conn
    conn.connect()
  File "/usr/lib/python3.12/site-packages/urllib3/connection.py", line 419, in connect
    self.sock = ssl_wrap_socket(
                ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/util/ssl_.py", line 462, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/util/ssl_.py", line 504, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/ssl.py", line 455, in wrap_socket
    return self.sslsocket_class._create(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/ssl.py", line 1041, in _create
    self.do_handshake()
  File "/usr/lib/python3.12/ssl.py", line 1319, in do_handshake
    self._sslobj.do_handshake()
TimeoutError: _ssl.c:993: The handshake operation timed out
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 667, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 802, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/util/retry.py", line 552, in increment
    raise six.reraise(type(error), error, _stacktrace)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/packages/six.py", line 770, in reraise
    raise value
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 716, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
[... 14 lines cut ...]
  File "/app/pytedapi/__init__.py", line 147, in connect
    self.reconnect()
  File "/app/pytedapi/__init__.py", line 169, in reconnect
    self.get_din(force=True)
  File "/app/pytedapi/__init__.py", line 267, in get_din
    r = self.request("tedapi/din", force=force)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 212, in request
    r = requests.get(url,
        ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 713, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPSConnectionPool(host='192.168.91.1', port=443): Read timed out. (read timeout=5)`

F1p, 2025-04-13:
I'm prepared with a Pi & PoE and using standalone from @sbman, but this isn't good news if connection to the local AP also doesn't work

I started to look at ESP32, but going to hold off putting the effort into converting the TEDAPI to cpp until a solution going forward is hopefully sorted

slyglif, 2025-04-13:
I'll start testing and trying to figure it out in a little over a month.  Hopefully it works and doesn't require flipping the power switch or something.

Fenix-HA, 2025-04-14:
Morning guys! Today I tried again with HA connected throught PW's WiFi...started plug-in, no error...refreshed mqtt device and...BOOM!!! Now I can see all sensor again, I'm so happy like a baby lol. Hope this is first step in right direction

Fenix-HA, 2025-04-16:
So a little update, I've configured a repeater to extend PW's WiFi. HA is connected to repeater WiFi. All seems to be working great, will be testing for the next couple of days...

jamesgreeley, 2025-04-18:
> I have an image that is setup for use in standalone.
>
> You can see a sample docker compose file here: [...]

Sadly I get the following error using your docker-compose example

[image]

my tedapi_password is set
[image]

shellhouse4, 2025-04-21:
> I've also managed to resolve the issue by having my Pi simultaneously connected to the PW3 WiFi and ethernet to my router (fortunately the Pi is in range of the PW3).
> 
> It's still necessary to have the static route active in order for the MQTT connection to use the correct network interface, with the following bizarre looking config:
> 
> ```
> - nexthop: 192.168.91.1
>   network: 192.168.91.1/32
> ```
> 
> Tesla is giving me serious buyer's remorse at the moment!

Similar experience here.  I set up another mini PC (x86) with Lubuntu, configured a wireless adapter over CLI,/SSH/wired and then set the wireless connection in HA under system -> network -> tab for wireless network.  Running wired for main HA data and wireless just for PW3 network.

Here's a quick article on setting up HA on Ubuntu, for anyone interested:
https://pimylifeup.com/ubuntu-home-assistant/

Would have preferred to use MX Linux in hindsight after testing MX, Mint, and Lubuntu on different devices, and such a distro was preferred over HAOS in order to accommodate other use cases besides HA.

Also had to use the static route plugin with the same details as above.

obrendino, 2025-04-26:
Mine broke yesterday like everyone elses.  Connected to PW3 via wifi and my network via ethernet changed the static route as others have done but no success.  I'm running a Virtualbox NUC.  Is there something else I need to do?
Also have almost got Tesla Fleet working but the API doesn't like hosting on Fleetkey.cc.  Trying get access to the switches.

slyglif, 2025-04-26:
> changed the static route as others have done

Once your HA machine is connected directly to the PW WiFi you shouldn't need a static route and you can just remove the static routes add-on and reboot the HA server so it flushes the route table.

Muzach, 2025-04-28:
I had a spare AirPort Express so wiped that and configured it to join the TeslaPW_XXXXX network and then plugged it into my router with a static IP and set up a route from 192.168.91.1/32 with nexthop 192.168.1.154 (the Ethernet of the AirPort Express).  But I’m still getting the unable to connect message below.  Anyone spot what I’m missing in building a bridge to the Tesla WiFi?  

> ERROR - Unable to connect to Powerwall Gateway: 192.168.91.1
2025-04-27T165343.138 - pytedapi - ERROR - Please verify your your host has a route to the Gateway.
2025-04-27T165343.138 - __main__ - CRITICAL - Exiting: Unable to connect to Powerwall

slyglif, 2025-04-28:
> I had a spare AirPort Express so wiped that and configured it to join the TeslaPW_XXXXX network and then plugged it into my router with a static IP and set up a route from 192.168.91.1/32 with [...]

I'm not sure exactly what your network looks like, but the 2 critical parts for communication to work are:
1. You have to connect to the PW via the TeslaPW_XXXXX network
2. The source IP for the communications has to be on the 192.168.91.0/24 network

(2) means that if you are doing some kind of routing, the device that is on the Tesla WiFi needs to NAT so the PW won't reject the https connection.  If you aren't doing any routing, and it's just a true transparent bridge, then your HA server should have a 192.168.91.x/24 IP itself, and no static routes would be needed.

obrendino, 2025-04-28:
I get the same error.  I have a 10.x.x.x network which the PW and homeassistant are connected to, but have got the homassistant NUC connected to 10.x.x.x via ethernet.  Can't seem to get onto the 192.168.91.1.

slyglif, 2025-04-28:
> I get the same error. I have a 10.x.x.x network which the PW and homeassistant are connected to, but have got the homassistant NUC connected to 10.x.x.x via ethernet. Can't seem to get onto the 192.168.91.1.

The only way to get the 192.168.91.1 IP to work is if you have DIRECT access to the TeslaPW_XXXXX WiFi network.  For the vast majority of people there can not be a wired network involved in HA talking to the PW (doing so requires a more advanced network setup and much more knowledge of how routing works).

slyglif, 2025-04-28:
I've updated the README to address the new requirements.  Because this is a requirement from the Powerwall itself, and there has been no workaround found by any of the 3 projects that try to use the TEDAPI, I'm closing this as a won't fix.

closed 2025-04-28 by slyglif

slyglif, 2025-05-12:
> I know this is closed but it's super frustrating that Tesla is closing access to the local api like this. Just paid a lot for PW3 and now they want to force me to pay for the fleet api to get metrics that already live on my local network 😠

Unless they just changed it, Tesla doesn't currently charge for fleet api access to powerwall data, only vehicle data.

cby016, 2025-05-12:
Ok I didn't know it was free for powerwall, I'll look into it a bit more.  Even still, relying on the cloud to get metrics from a local device doesn't seem right to me.