slyglif/powerwall3mqtt #37: Frequent restarts

bjquinn, 2025-04-06:
I'm getting frequent restarts.  In between restarts, I do seem to be getting all data points correctly.  Is there any other info I could provide that would be helpful?

Supervisor logs
`
2025-04-06 13:53:44.010 INFO (MainThread) [supervisor.resolution.checks.base] Run check for trust/supervisor
2025-04-06 13:53:48.732 WARNING (MainThread) [supervisor.homeassistant.websocket] Connection is closed
2025-04-06 13:54:23.001 INFO (MainThread) [supervisor.resolution.checks.base] Run check for pwned/addon
2025-04-06 13:54:37.945 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2025-04-06 13:54:47.922 INFO (MainThread) [supervisor.jobs] 'CheckAddonPwned.run_check' blocked from execution, no supervisor internet connection
2025-04-06 13:54:50.822 INFO (MainThread) [supervisor.resolution.checks.base] Run check for multiple_data_disks/system
2025-04-06 13:54:52.193 INFO (MainThread) [supervisor.resolution.checks.base] Run check for docker_config/system
2025-04-06 13:54:54.113 INFO (MainThread) [supervisor.resolution.checks.base] Run check for disabled_data_disk/system
2025-04-06 13:54:54.114 INFO (MainThread) [supervisor.resolution.check] System checks complete
2025-04-06 13:54:54.114 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state running
2025-04-06 13:55:27.687 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2025-04-06 13:56:09.571 INFO (SyncWorker_6) [supervisor.docker.manager] Cleaning hassio_observer application
2025-04-06 13:56:43.677 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
2025-04-06 13:56:43.766 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state running
2025-04-06 13:56:43.766 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
2025-04-06 13:56:49.068 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
2025-04-06 13:57:24.712 INFO (MainThread) [supervisor.docker.observer] Starting Observer ghcr.io/home-assistant/aarch64-hassio-observer with version 2025.02.0 - 172.30.32.6
2025-04-06 13:58:22.588 INFO (SyncWorker_3) [supervisor.docker.manager] Cleaning addon_core_matter_server application
2025-04-06 13:58:24.918 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-matter-server with version 7.0.0
2025-04-06 13:58:37.141 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_matter_server
2025-04-06 13:58:45.309 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-04-06 13:59:23.834 INFO (SyncWorker_2) [supervisor.docker.manager] Cleaning addon_core_matter_server application
2025-04-06 13:59:27.069 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-matter-server with version 7.0.0
2025-04-06 13:59:38.988 INFO (MainThread) [supervisor.api.middleware.security] /network/info access from core_matter_server
2025-04-06 14:00:43.725 WARNING (MainThread) [supervisor.addons.addon] Watchdog found addon Powerwall3 to MQTT is failed, restarting...
2025-04-06 14:00:43.783 INFO (SyncWorker_5) [supervisor.docker.manager] Cleaning addon_e5f48848_powerwall3mqtt application
2025-04-06 14:00:50.972 INFO (MainThread) [supervisor.docker.addon] Starting Docker add-on e5f48848/aarch64-addon-powerwall3mqtt with version 0.3.8
`

powerwall3mqtt logs
`
[13:46:45] INFO: Preparing to start...
[13:46:49] INFO: Starting Powerwall3MQTT...
[13:47:09] INFO: Preparing to start...
[13:47:11] INFO: Starting Powerwall3MQTT...
[13:51:35] INFO: Preparing to start...
[13:51:37] INFO: Starting Powerwall3MQTT...
[14:00:51] INFO: Preparing to start...
[14:00:53] INFO: Starting Powerwall3MQTT...
[14:05:18] INFO: Preparing to start...
[14:05:21] INFO: Starting Powerwall3MQTT...
`

slyglif, 2025-04-06:
Try turning on debug logging for in the configuration.  Also, I don't have watchdog support working yet, so if you have that enabled, turn that off as it would likely restart the add-on even when it is working correctly.

bjquinn, 2025-04-07:
Ok, I turned off watchdog, restarted it, and it ran for a couple hours.  I can't see if there's anything else useful in there, it seems to be the below repeated over and over.  This is the last series of events.  One difference is that I see the "Using Cached din" line is repeated twice, and I don't see that in previous iterations.

```
2025-04-06T194959.868 - pytedapi - DEBUG - Using Cached din
2025-04-06T195008.811 - pytedapi - DEBUG - Status: {'control': {'alerts': {'active': ['SystemConnectedToGrid', 'NetLinkStatusChange', 'FWUpdateSucceeded']}, 'batteryBlocks': [{'din': '1707000-11-J--TG124061002APR', 'disableReasons': []}], 'islanding': {'contactorClosed': True, 'customerIslandMode': 'BACKUP', 'gridOK': True, 'microGridOK': True}, 'meterAggregates': [{'location': 'BATTERY', 'realPowerW': 1839}, {'location': 'SITE', 'realPowerW': 28.699996948242188}, {'location': 'LOAD', 'realPowerW': 1853.1500091552737}, {'location': 'SOLAR', 'realPowerW': 0}, {'location': 'SOLAR_RGM', 'realPowerW': 0}, {'location': 'GENERATOR', 'realPowerW': 0}, {'location': 'CONDUCTOR', 'realPowerW': 0}], 'pvInverters': [], 'siteShutdown': {'isShutDown': False, 'reasons': []}, 'systemStatus': {'nominalEnergyRemainingWh': 10300, 'nominalFullPackEnergyWh': 14400}}, 'esCan': {'bus': {'ISLANDER': {'ISLAND_AcMeasurements': {'ISLAND_FreqL1_Load': 59.98, 'ISLAND_FreqL1_Main': 59.98, 'ISLAND_FreqL2_Load': 59.98, 'ISLAND_FreqL2_Main': 59.98, 'ISLAND_FreqL3_Load': 0, 'ISLAND_FreqL3_Main': 0, 'ISLAND_GridState': 'ISLAND_GridState_Grid_Compliant', 'ISLAND_VL1N_Load': 125.5, 'ISLAND_VL1N_Main': 125.5, 'ISLAND_VL2N_Load': 125, 'ISLAND_VL2N_Main': 125, 'ISLAND_VL3N_Load': 0, 'ISLAND_VL3N_Main': 0, 'isComplete': True, 'isMIA': False, 'lastRxTime': '2025-04-06T19:50:08-05:00'}, 'ISLAND_GridConnection': {'ISLAND_GridConnected': 'ISLAND_GridConnected_Connected', 'isComplete': True}}, 'MSA': {'METER_Z_AcMeasurements': {'METER_Z_CTA_I': 0, 'METER_Z_CTA_InstReactivePower': 0, 'METER_Z_CTA_InstRealPower': 0, 'METER_Z_CTB_I': 0, 'METER_Z_CTB_InstReactivePower': 0, 'METER_Z_CTB_InstRealPower': 0, 'METER_Z_VL1G': 0, 'METER_Z_VL2G': 0, 'isMIA': True, 'lastRxTime': '0001-01-01T00:00:00Z'}, 'MSA_InfoMsg': {'MSA_appGitHash': [0, 0, 0, 0, 0, 0, 0], 'MSA_assemblyId': 0, 'isMIA': True}, 'MSA_Status': {'lastRxTime': '0001-01-01T00:00:00Z'}, 'packagePartNumber': '', 'packageSerialNumber': ''}, 'PINV': [{'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridState': 'Grid_Uncompliant', 'PINV_Pout': 0, 'PINV_State': 'PINV_Off', 'PINV_Vout': 0, 'isMIA': True}, 'alerts': {'active': [], 'isComplete': False, 'isMIA': True}}, {'PINV_AcMeasurements': {'PINV_VSplit1': 0, 'PINV_VSplit2': 0, 'isMIA': True}, 'PINV_PowerCapability': {'PINV_Pnom': 0, 'isComplete': False, 'isMIA': True}, 'PINV_Status': {'PINV_Fout': 0, 'PINV_GridStat [... cut ...] , "calibration": "OFF", "commission_date": "2024-05-06T11:13:04-05:00", "grid_power": 28.7, "grid_status": "ON", "inverter_capacity": 11520.0, "load_power": 1853.15, "solar_power": 0, "real_power_config_limited": "OFF"}
2025-04-06T195034.597 - __main__ - DEBUG - Processing update from timing_loop
2025-04-06T195037.922 - pytedapi - DEBUG - Get Firmware Version from Powerwall
2025-04-06T195043.565 - pytedapi - DEBUG - Using Cached din
2025-04-06T195043.565 - pytedapi - DEBUG - Using Cached din
```

slyglif, 2025-04-07:
Could you paste or attach the first and last 100 lines of the logs?

bjquinn, 2025-04-07:
Attached entire log file.  Thanks for your help, this project is awesome.  I'm super excited to see what I can do hooking in my solar/battery to automations!

[e5f48848_powerwall3mqtt_2025-04-07T15-55-22.750Z.log](https://github.com/user-attachments/files/19635521/e5f48848_powerwall3mqtt_2025-04-07T15-55-22.750Z.log)

slyglif, 2025-04-07:
1GB is definitely on the low side.  There have been reports of memory problems since HAOS 15.1 released.  It is possible that the log is just cluttered with watchdog restarts, but some of the ones where it restarted back to back without any output still don't seem right, and feel more like the host is out of memory and the kernel is OOM killing processes / containers.  Do you have any way to monitor resource utilization on the host (especially memory)?

closed 2025-04-07 by bjquinn

bjquinn, 2025-04-09:
Just to follow up here, it appears the upgrade to an 8GB raspberry pi 5 worked.  It's using 0.9GB of RAM, so it must have been juuuust right at the line.  It's been working for about 12 hours with no issues.  I don't think I ever had it run that long previously.

slyglif, 2025-04-09:
Glad it was an easy fix!