slyglif/powerwall3mqtt #40: Powerwall3mqtt startup error

devangapatel, 2025-04-30:
I had the static routes addon and powerwall3mqtt addon configured and working in my HA. It worked great for a few months and broke last weekend. Upon restarting everything, I am getting the following in the logs:

[14:20:13] INFO: Preparing to start...
[14:20:15] INFO: Starting Powerwall3MQTT...
2025-04-30T142017.651 - __main__ - ERROR - (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')
Traceback (most recent call last):
  File "/app/powerwall3mqtt.py", line 361, in <module>
    app.run()
  File "/app/powerwall3mqtt.py", line 301, in run
    tedapi = pytedapi.TeslaEnergyDeviceAPI(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 129, in __init__
    self.connect()
  File "/app/pytedapi/__init__.py", line 147, in connect
    self.reconnect()
  File "/app/pytedapi/__init__.py", line 169, in reconnect
    self.get_din(force=True)
  File "/app/pytedapi/__init__.py", line 267, in get_din
    r = self.request("tedapi/din", force=force)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 216, in request
    self.check_http_response(r)
  File "/app/pytedapi/__init__.py", line 186, in check_http_response
    raise exceptions.TEDAPIAccessDeniedException()
pytedapi.exceptions.TEDAPIAccessDeniedException: (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')

I dont believe I have changed any password for the the Tesla PW3. Any insight on debugging or troubleshooting this would be appreciated.

slyglif, 2025-04-30:
Tesla's latest firmware updates prevent all access via the Ethernet connection, only allowing access via the WiFi connection, and the source IP must be from the 192.168.91.0/24 network.  It effectively means the only easy way to make it work is to not use the static routes anymore and you need to join your HA server to the TeslaPW_XXXXX WiFi network directly.

closed 2025-04-30 by slyglif

devangapatel, 2025-04-30:
> Tesla's latest firmware updates prevent all access via the Ethernet connection, only allowing access via the WiFi connection, and the source IP must be from the 192.168.91.0/24 network. It [...]

Thanks for this insight. So my HA server [VM on Synology NAS] is connected to the main wifi network in my home. How can I also connect it to the Tesla PW3 wifi simultaneously? Any help here would be appreciated.

slyglif, 2025-04-30:
You'd need to do one of the following:
- add a second WiFi card somehow (I'm not familiar with Synology devices, so don't know if that is even an option)
- get a WiFi bridge device (that can act as a WiFi client, not access point), plug it into the Ethernet port on the NAS, and join the bridge to the Tesla WiFi network
- build some kind of host (like a Raspberry Pi) to connect to the Tesla WiFi and your main network via Ethernet, and either run this as a container on that host, or setup that host to act as a NAT gateway device

Unfortunately setting up any of these is well beyond the help/advice I can provide.  Tesla has basically decided they don't want people directly monitoring their equipment and has taken explicit steps to prevent people from doing it.  I actually wouldn't be surprised if they restrict it even further so it is difficult to do on their WiFi network as well, but that's just speculation.