slyglif/powerwall3mqtt #42: Sudden issues with acessing PW3 TEDAPIAccessDeniedException

meow4agreencard, 2025-05-23:
The add-on has been humming along for the last few months without hiccups. As of last night night, I'm getting a password exception.

Obviously, I confirmed that the password is still correct. I haven't changed anything on my side.
I confirmed that I can login to the GW with NetZero using the same password.
I confirmed that HA still has a route to 192.168.91.1 via ping.

Not sure what else could be causing this. Is the GW rejecting the add-on from logging in?

```[08:20:20] INFO: Preparing to start...
[08:20:21] INFO: Starting Powerwall3MQTT...
2025-05-23T082021.491 - __main__ - DEBUG - Runtime config:
2025-05-23T082021.491 - __main__ - DEBUG - config['log_level'] = 'DEBUG'
2025-05-23T082021.491 - __main__ - DEBUG - config['mqtt_base_topic'] = 'homeassistant'
2025-05-23T082021.491 - __main__ - DEBUG - config['mqtt_ca'] = ''
2025-05-23T082021.492 - __main__ - DEBUG - config['mqtt_cert'] = ''
2025-05-23T082021.492 - __main__ - DEBUG - config['mqtt_host'] = 'core-mosquitto'
2025-05-23T082021.492 - __main__ - DEBUG - config['mqtt_key'] = ''
2025-05-23T082021.492 - __main__ - DEBUG - config['mqtt_password'] = 'XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX'
2025-05-23T082021.492 - __main__ - DEBUG - config['mqtt_port'] = '1883'
2025-05-23T082021.492 - __main__ - DEBUG - config['mqtt_ssl'] = 'False'
2025-05-23T082021.492 - __main__ - DEBUG - config['mqtt_username'] = 'addons'
2025-05-23T082021.492 - __main__ - DEBUG - config['mqtt_verify_tls'] = 'False'
2025-05-23T082021.492 - __main__ - DEBUG - config['tedapi_host'] = '192.168.91.1'
2025-05-23T082021.492 - __main__ - DEBUG - config['tedapi_password'] = 'XXXXXXXXXX'
2025-05-23T082021.492 - __main__ - DEBUG - config['tedapi_poll_interval'] = '5'
2025-05-23T082021.492 - __main__ - DEBUG - config['tedapi_report_vitals'] = 'True'
2025-05-23T082021.493 - __main__ - DEBUG - MQTT will set on 'powerwall3mqtt-15/will' to 'b'offline''
2025-05-23T082021.494 - pytedapi - DEBUG - Testing Connection to Powerwall Gateway: 192.168.91.1
2025-05-23T082021.542 - pytedapi - DEBUG - Detected Powerwall 3 Gateway
2025-05-23T082021.542 - pytedapi - DEBUG - Fetching din from Powerwall...
2025-05-23T082021.600 - __main__ - ERROR - (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')
Traceback (most recent call last):
  File "/app/powerwall3mqtt.py", line 361, in <module>
    app.run()
  File "/app/powerwall3mqtt.py", line 301, in run
    tedapi = pytedapi.TeslaEnergyDeviceAPI(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 129, in __init__
    self.connect()
  File "/app/pytedapi/__init__.py", line 147, in connect
    self.reconnect()
  File "/app/pytedapi/__init__.py", line 169, in reconnect
    self.get_din(force=True)
  File "/app/pytedapi/__init__.py", line 267, in get_din
    r = self.request("tedapi/din", force=force)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 216, in request
    self.check_http_response(r)
  File "/app/pytedapi/__init__.py", line 186, in check_http_response
    raise exceptions.TEDAPIAccessDeniedException()
pytedapi.exceptions.TEDAPIAccessDeniedException: (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')```

slyglif, 2025-05-23:
Use NetZero to check what firmware version the PW is now using.  I suspect it updated to 25.10.1 or newer.  If that's the case, the problem is because Tesla changed their requirements and it is no longer possible to use the TEDAPI over wired ethernet, and the only way is to connect to the TeslaPW_XXXXX wifi network and access it from a 192.168.91.x IP address.

meow4agreencard, 2025-05-23:
It's running 25.10.4. My PW is connected to my router via WiFi and I have a route to 192.168.91.1 from the HA network.
You're saying that I will now have to somehow get my HA IP to show up as 91.x to the PW?

slyglif, 2025-05-23:
Yes, but it's more than that.  It isn't that it just has to show up as 91.x to the PW, it's that it needs to show up as that IP on the TeslaPW_XXXXX network, because that's where the PW will try to send the return traffic (even though that isn't where the traffic came from).

If your HA server has an idle WiFi NIC, you could have HA join the PW's WiFi network, which would automatically give it a 192.168.91.X IP.  You'd leave the wired ethernet connection alone so you can continue to use HA.

nrlcode, 2025-05-23:
Is it possible to still access it via static routes? I'm able to communicate with it via the 192.168.91.X address, ping works for example. However, I'm not able to authenticate. 

```
curl -k --user "Tesla_Energy_Device:XXXXXXXXXX" https://192.168.91.1/tedapi/din
{"code":403,"error":"Unable to GET to resource","message":"User does not have adequate access rights"}
```

slyglif, 2025-05-23:
> Is it possible to still access it via static routes? I'm able to communicate with it via the 192.168.91.X address, ping works for example. However, I'm not able to authenticate.

Unfortunately no, Tesla added a check to make sure it can't authenticate until it comes in through the TeslaPW_XXXXX wifi network.  It appears they REALLY don't want people to collect metrics locally.

closed 2025-05-24 by slyglif

F1p, 2025-05-29:
@slyglif - No discussions option on this repo, but wanted to ask how NetZero connectors and the control options available through it are available but not yet built into your repo?

slyglif, 2025-05-29:
Yep, I still have Issue #6 open for that reason.  So far it looks like doing so requires pairing the client (in this case this application) with the powerwall, which seems to require communicating with Tesla's private cloud API (not fleet API) to get a token that can be presented to the PW for authentication.  Without a way to figure out the private API to their cloud we can't get the token.

F1p, 2025-05-29:
> Yep, I still have Issue [#6](https://github.com/slyglif/powerwall3mqtt/issues/6) open for that reason. So far it looks like doing so requires pairing the client (in this case this application) with [...]

Great, smart but annoying 🤔

Marsi90, 2025-05-30:
Hi @slyglif :)
First of all, you really make a great job. 

I am struggling with this issue, too. But I didn‘t unterstand what is the solution? Can you explain it a little bit more detailed please? I am not an expert with networks :-/

Kind regards and a nice day :)

Marsi90, 2025-05-30:
Hey @slyglif  and @meow4agreencard,

thx for helping. I tried the first option and connected my HA to the PW Network, deleted static routes, restarted HA. Everything is working again :) 
I hope it stays that way.. what do you think? Can Tesla delete this option again?

Again thank you and a nice evening :)

carbon198, 2025-05-30:
I don’t think that they will. More likely they would change how the API
works, but hopefully that won’t happen unless they’re giving us a much
better API.

slyglif, 2025-05-30:
I don't believe they would kill this method completely, as it would make provisioning in areas without cellular service next to impossible.  As carbon198 says, they could change the API, but they would only do that because they have a new one we should use, or because they're being spiteful and want to somehow monetize the data like they are now doing with car level stats through the FleetAPI.

sbman, 2025-06-10:
This version finally hit my set of Powerwall 3 units.  I added a USB Wifi adapter to my server, configured it to join my powerwall Wifi (they are in the same room) and then passed that network through to my docker container and it is all running great.  The only oddity I have is that I can only see 1 string data on one powerwall, the other 2 strings connected to the other powerwall do not show up.

sbman, 2025-06-10:
It looks like I can only see string data on the 'primary' powerwall.  The secondary powerwall has two more strings attached to it and that data isn't being reported to MQTT, and it does not appear to be in the JSON, just has all blank values.

sbman, 2025-06-10:
I do get all other info from both powerwalls, just not the string data.

sbman, 2025-06-11:
Looks like the entities for the other strings were disabled in Home Assistant so I just was not seeing them, but they are all showing up now.