slyglif/powerwall3mqtt #43: TEDAPI Access Denied

Vadimvs7, 2025-06-02:
Hi,
I have an issue/observation to report. I have been using the plugin for over a month (thanks!!!) and it has been working well until recently. But now it seems like it has lost its ability to log in. I can ping PW3 IP x.x.91.1 from the HA host. Update interval is at 5 seconds and I did not change it. 
Needless to say, password was not changed, and I can still connect to the PW3 wi-fi network with the original password. 
Here's log output:
[00:16:42] INFO: Preparing to start...
[00:16:43] INFO: Starting Powerwall3MQTT...
2025-06-02T001645.115 - __main__ - ERROR - (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')
Traceback (most recent call last):
  File "/app/powerwall3mqtt.py", line 361, in <module>
    app.run()
  File "/app/powerwall3mqtt.py", line 301, in run
    tedapi = pytedapi.TeslaEnergyDeviceAPI(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 129, in __init__
    self.connect()
  File "/app/pytedapi/__init__.py", line 147, in connect
    self.reconnect()
  File "/app/pytedapi/__init__.py", line 169, in reconnect
    self.get_din(force=True)
  File "/app/pytedapi/__init__.py", line 267, in get_din
    r = self.request("tedapi/din", force=force)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 216, in request
    self.check_http_response(r)
  File "/app/pytedapi/__init__.py", line 186, in check_http_response
    raise exceptions.TEDAPIAccessDeniedException()
pytedapi.exceptions.TEDAPIAccessDeniedException: (TEDAPIAccessDeniedException(...), 'Access Denied: Check your Gateway Password')

slyglif, 2025-06-02:
Can you confirm the HA server is directly connected to the TeslaPW_XXXXX wifi network from the Powerwall leader?  See the first item in the readme and/or the Pre-Reqs section of the readme.

Vadimvs7, 2025-06-04:
Right, I don't satisfy these Pre-Reqs. PW3 is on my local wi-fi network, I added a route on my HA host to connect to 192.168.91.1 via whatever IP PW3 has on my local LAN and it was working until 2 weeks ago.
I can still ping 192.168.91.1 from within HA OS, and PW3 communicates, but returns "wrong password". I am sure I was on a firmware at or later than 25.10.1 and it was working, but not any more. My current firmware version is 25.10.4
Oh well, I think it's the PW3 that is misbehaving, so I guess the issue is not with powerwall3mqtt.
Thank you for your help any way.

slyglif, 2025-06-04:
One of the versions in the 25.10.x range is where Tesla changed their system to actively prevent the local routing method from working, making it so it only works if you connect to the TeslaPW_XXXXX WiFi.  Unfortunately there is no known way around that at this point.

closed 2025-06-04 by slyglif