slyglif/powerwall3mqtt #51: [Errno 113] Host is unreachable

Magic73, 2025-09-02:
I'm getting a lot of errors like:
[Errno 113] Host is unreachable

I can't figure out the cause, although the WiFi connection to Tesla is stable (I have an addon that checks the connection status every 30 seconds, and it never detects WiFi disconnections).

I can recover the data (I have 60 seconds as the interval) but I have several missing data caused by this very problem.

The strange thing is that this problem only occurs with the string sensors (the Report Vitals feature is enabled), while the main recovery sensors never fail.

I have a powerwall 3, with firmware: 25.26.1 2c4bb00e

`  File "/app/powerwall3mqtt.py", line 259, in main_loop
    self.update(mqtt, tesla, True)
  File "/app/powerwall3mqtt.py", line 355, in update
    tesla.update()
  File "/app/hamqtt/devices.py", line 312, in update
    firmware = self.tedapi.get_firmware_version(details=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 610, in get_firmware_version
    r = self._tesla.post(
        ^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 241, in post
    r = requests.post(url,
        ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 115, in post
    return request("post", url, data=data, json=json, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 700, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fe5ab645b50>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
2025-09-02T161105.729 - __main__ - ERROR - HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fe5ab4d7050>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/util/connection.py", line 95, in create_connection
    raise err
  File "/usr/lib/python3.12/site-packages/urllib3/util/connection.py", line 85, in create_connection
    sock.connect(sa)
OSError: [Errno 113] Host is unreachable
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 716, in urlopen
    httplib_response = self._make_request(
                       ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 404, in _make_request
    self._validate_conn(conn)
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 1061, in _validate_conn
    conn.connect()
  File "/usr/lib/python3.12/site-packages/urllib3/connection.py", line 363, in connect
    self.sock = conn = self._new_conn()
                       ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connection.py", line 186, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x7fe5ab4d7050>: Failed to establish a new connection: [Errno 113] Host is unreachable
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 667, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/connectionpool.py", line 802, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/urllib3/util/retry.py", line 594, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fe5ab4d7050>: Failed to establish a new connection: [Errno 113] Host is unreachable'))
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/app/powerwall3mqtt.py", line 259, in main_loop
    self.update(mqtt, tesla, True)
  File "/app/powerwall3mqtt.py", line 355, in update
    tesla.update()
  File "/app/hamqtt/devices.py", line 312, in update
    firmware = self.tedapi.get_firmware_version(details=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 610, in get_firmware_version
    r = self._tesla.post(
        ^^^^^^^^^^^^^^^^^
  File "/app/pytedapi/__init__.py", line 241, in post
    r = requests.post(url,
        ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 115, in post
    return request("post", url, data=data, json=json, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/requests/adapters.py", line 700, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='192.168.91.1', port=443): Max retries exceeded with url: /tedapi/v1 (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fe5ab4d7050>: Failed to establish a new connection: [Errno 113] Host is unreachable'))`

slyglif, 2025-09-02:
Hmm.  Any chance you could check the host logs for anything in dmesg or something from the wifi supplicant?  I want if the PW is too slow to respond to WiFi itself, but not slow enough that the connection fully drops.

Magic73, 2025-09-02:
as example, this is the battery history:

[image]

and this is the PV String A, for the same period:

[image]

slyglif, 2025-09-02:
I suspect that is the issue (likely caused because the Tesla WiFi has to switch between hostap mode and client mode, which is unavoidable).  I believe the same is happening to me using the WiFi bridge I mentioned, but since it's a dump bridge I can't see any logs, just the timeouts caused by it.

Magic73, 2025-09-02:
I'm not confident with python, but with the help of Gemini, does it would works? (post method of __init__.py)

`
import tenacity

def post(self, path, force=False, headers=None, data=None):
        """
        Make an HTTP POST request to the Powerwall Gateway, converting
        some HTTP status codes to exceptions
        Parameters:
            path (str): The URI path
            force (bool): Force a query from the API, default false
            headers (dict): Passed through to requests, default None
            data (dict): Passed through to requests, default None
        Returns:
            requests.Response: The HTTP resposne
        Raises:
            TEDAPIRateLimitedException
            TEDAPIRateLimitingException
            TEDAPIAccessDeniedException
            TEDAPIException
        """
        if not force and self._pwcooldown > time.perf_counter():
            raise exceptions.TEDAPIRateLimitedException()
        with self._api_lock:
            url = f"https://{self._gw_ip}/{path}"

        @tenacity.retry(
            wait=tenacity.wait_exponential(multiplier=2, min=2, max=32),
            stop=tenacity.stop_after_attempt(5),
            retry=(tenacity.retry_if_exception_type(requests.exceptions.ConnectionError) |
                   tenacity.retry_if_exception_type(requests.exceptions.Timeout))
        )
        def _make_request():
            return requests.post(url,
                verify=False,
                auth=('Tesla_Energy_Device', self._gw_pwd),
                headers=headers,
                data=data,
                timeout=self._timeout)

        try:
            r = _make_request()
        except (requests.exceptions.ConnectionError, requests.exceptions.Timeout) as e:
            raise exceptions.TEDAPIException("Failed to reach host after multiple retries") from e

            self.check_http_response(r)
            self._pwcooldown = time.perf_counter()
            return r

`

slyglif, 2025-09-02:
Right now the retries are built into urllib, but I'll check how easily I can add exponential backoff retries.

I did give up on using the API and helpers for energy stats, and instead setup the Tesla Fleet integration for that info.  Now I just use this for spot power info to feed into the Power Flow card.  Hopefully someday Tesla will see the light and provide official local API support (which I would immediately start using for this integration), but I doubt it will happen.

Magic73, 2025-11-17:
yes, I updating my fork...